tango2590/Hughes-Power-Watchdog #3: Error when updating board after ESPHome 2022.9.4 update

phurth, 2022-10-07:
After updating ESPHome from 2022.9.3 to 2022.9.4, when trying to update the board with the Power Watchdog code I'm getting a failure with the following errors:
```
INFO Reading configuration /config/esphome/power-watchdog.yaml...
INFO Updating https://github.com/spbrogan/esphome@PolledSensor
INFO Generating C++ source...
INFO Compiling app...
Processing power-watchdog (board: nodemcu-32s; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
Library Manager: Installing Hash
INFO Installing Hash
Library Manager: Installing ESP8266WiFi
INFO Installing ESP8266WiFi
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 1.0
|-- FS @ 1.0
|-- Update @ 1.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 1.1.0
|-- ESPmDNS @ 1.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
Compiling /data/power-watchdog/.pioenvs/power-watchdog/src/esphome/components/hughes_power_watchdog/hughes_power_watchdog.cpp.o
Compiling /data/power-watchdog/.pioenvs/power-watchdog/src/esphome/core/util.cpp.o
src/esphome/components/hughes_power_watchdog/hughes_power_watchdog.cpp: In member function 'virtual void esphome::hughes_power_watchdog::HughesPowerWatchdog::gattc_event_handler(esp_gattc_cb_event_t, esp_gatt_if_t, esp_ble_gattc_cb_param_t*)':
src/esphome/components/hughes_power_watchdog/hughes_power_watchdog.cpp:138:61: error: 'class esphome::ble_client::BLEClient' has no member named 'gattc_if'
           esp_ble_gattc_register_for_notify(this->parent()->gattc_if, this->parent()->remote_bda, chr->handle);
                                                             ^
src/esphome/components/hughes_power_watchdog/hughes_power_watchdog.cpp:138:87: error: 'class esphome::ble_client::BLEClient' has no member named 'remote_bda'
           esp_ble_gattc_register_for_notify(this->parent()->gattc_if, this->parent()->remote_bda, chr->handle);
                                                                                       ^
src/esphome/components/hughes_power_watchdog/hughes_power_watchdog.cpp:146:52: error: 'class esphome::ble_client::BLEClient' has no member named 'conn_id'
       if (param->notify.conn_id != this->parent()->conn_id || param->notify.handle != this->handle)
                                                    ^
*** [/data/power-watchdog/.pioenvs/power-watchdog/src/esphome/components/hughes_power_watchdog/hughes_power_watchdog.cpp.o] Error 1
========================== [FAILED] Took 3.27 seconds ==========================
```
Past ESPHome updates have been uneventful, but I know HA has been making a bunch of BT changes lately.

Muskrat775, 2022-10-19:
> After updating ESPHome from 2022.9.3 to 2022.9.4, when trying to update the board with the Power Watchdog code I'm getting a failure with the following errors:
>
> ```
> INFO Reading configuration [...]

Same errors in ESPHome 2022.10.0

kwe1009, 2022-10-19:
I am having the exact same issue.

tango2590, 2022-10-26:
The PolledSensor component was updated to fix this issue. This should be working again in 2022.10.1.

closed 2022-10-26 by tango2590