tango2590/Hughes-Power-Watchdog #4: Feature Request - Unit of Measure

Gliebig, 2023-01-27:
I tried the BLE scanner from bluepixel technologoes and the Nordic nRF connect scanners on my Samsung Galaxy Note 9. It was quite easy to find the MAC address once I was able to isolate the device and know what I was looking for. I have the 30 amp version of the Hughes Power Watchdog. Not sure if that is why the prefix is different. Mine begins with PMS... (maybe single line?) Once I got that figured out and reconfigured your yaml it worked straight away! I'm no expert, but do enjoy solving problems. I agree the energy reset feature would be great. Also, there are limits that can be set on my watchdog including Max Energy (kWh), Min and Max Voltages, and Max Amps. Those limits aren't as important with the integration into HA, but if they were able to be synced it might prove helpful when the ESP32 is not connected to it. Just a thought.

Since these devices are usually used on RV's, it might be helpful in your YAML to include the unit_of_measurement. I also monitor dc volts and all my sensor data is stored in an InfluxBD and graphed with Grafana. The "V" identifier isn't descriptive enough pulling out voltage measurements. Just might help out those that just copy and paste.  Great job!

````
sensor:`
  - platform: hughes_power_watchdog
    ble_client_id: power_watchdog
    voltage_line_1:
      name: "Watchdog Voltage Line 1"
      unit_of_measurement: "Vac"
      accuracy_decimals: 0
    current_line_1:
      name: "Watchdog Current Line 1"
      unit_of_measurement: "A"
      accuracy_decimals: 1
````
     
I use the following standard:
      unit_of_measurement: "A" Amps
      unit_of_measurement: "Vac" Volts Alternating Current
      unit_of_measurement: "Vdc: Volts Direct Current

tango2590, 2023-01-28:
Strange that your device was identified with "PWS". Which model do you have? This was tested with the PWD50-EPO-H, purchased in 2021. Could be a difference between 30/50amp models or portable/hardwired versions. I also suspect you may have a newer unit, as mine does not offer those limits. I'll reach out to the Smart RV community to see if those limits can be added.

I believe the `unit_of_measurement`, `device_class`, and `state_class` are all done in the ESPHome component. I do remember we were having some issues at the beginning getting these sensors to correctly appear in the Energy dashboard. The ESP32 was also having memory issues when we added units to every sensor on the 50amps. I'll look in to this again.

spbrogan, 2023-01-28:
Here is how we have it configured in esphome. 
https://github.com/spbrogan/esphome/blob/7d89e91a184b429fa9eaf7aa1faecf0fdbe043b9/esphome/components/hughes_power_watchdog/sensor.py#L62

But looking at the code i don't see ESPHOME having another option.  
https://github.com/spbrogan/esphome/blob/7d89e91a184b429fa9eaf7aa1faecf0fdbe043b9/esphome/const.py#L893

And the docs
https://esphome.io/components/sensor/index.html?highlight=unit_of_measurement

Gliebig, 2023-01-29:
Yes I have Power Watchdog Surge Protector with Bluetooth - Portable - 30 Amp. I'm guessing the PWS where S means single line and PWD where D means dual line? I purchased mine in 2020. I don't have a dual line surge protector available to verify. I can look for one this summer during camping season to see if I can confirm. 

The weird thing with the phone app is shows the MAC address. It's the same address as the QR code sticker on the side of the surge protector, but it is not the correct address needed to make this integration work. This picture also shows the fields that can be set for the alerts that appear on the app. I found this is not very useful as the app needs to be connected to get the alerts and I don't have the app connected all the time when we are camping. This is the beauty of the integration into HA.

[image]

I looked through the ESPHome sensor docs you provided and concur it is not there, and it should be. There are so many of us using Grafana for making charts and having using the filter help tremendously when looking for sensor data.  Also there is a big difference between AC and DC volt systems and now with all the green energy systems out there they should have the correct unit_of_measurement.  On my RV I have mixed voltages and need to keep them straight.

What I did is changed your yaml  and reflashed the ESP32 and it's working perfectly without any errors with the correct unit_of_masurement.

spbrogan, 2023-01-29:
The alarms and notifications are features of the app not the device. This is why it requires a constant connection.  You could easily use home assistant, the companion mobile app, and automation to send a notification for similar events.

The QR code and that value that is shown on your app are an id that is in the report that is broadcast by the device.  When connecting with a new device the app uses the QR code to get that value and then checks all ble devices for a report with that value. Anyway, the integration doesn't use that. It uses the real MAC address.  I wonder if we should switch since the ble Mac isn't trivial to find for the average user.

[2023-01-23 16 44 02]<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F35844174%2F215339317-030bb066-00d9-4cfc-8c70-f65dd3616763.jpg&data=05%7C01%7C%7C3bc8fc84498d4834391e08db021502a8%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C638106061482964335%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=LzHp8J48H8WIs9e8fbWI7UjYfcSMwz9QBQE8Bf%2BmJN8%3D&reserved=0>

Gliebig, 2023-01-30:
It would make things a whole lot easier for people if they could use the QR MAC that appears in the app vs the real MAC since that code is right in front of the user no need to do any RF snooping. I don't know if that is possible. I really love having the data now streaming into my HA! Thanks so much for the integration!

tango2590, 2023-02-03:
I agree that using the QR code instead of the MAC address would be more straightforward. So here's where I'm at with it. My device's QR code is a device ID, not a MAC address. It's APMD24B81E309. If you drop the first letter (A), the resulting string is the name that's visible when searching for BLE devices (both on my Mac and my Android). From the looks of it, the BLE Client part of ESPHome requires a MAC address. Based on the docs, it doesn't look like it's possible to search via the device ID. Is this correct?