tango2590/Hughes-Power-Watchdog #5: Issue While Installing

mllockwood, 2023-03-24:
Installation of this continues to fail every time I attempt it.  I am not extremely well versed using ESPHome, I do admit.  Here's the steps I am using to install:

1.  I used a BLE sniffer and found my MAC address of my Hughes Power Watchdog.
2.  Installed the ESPHome addon in to Home Assistant.
3.  Open the web UI for ESPHome from within Home Assistant
4.  Click "Add Device" in the UI.
5.  Click "Continue" on the warning about ESPHome needing to be connected using a USB cable to add a device.
6.  Enter a name for the new configuration in the prompt.  I chose "Power Watchdog" and clicked Next.
7.  For Device Type, I choose ESP32.
8.  A box pops up stating "Configuration Created" and displays an Encryption Key.  I click "Skip"
9. I choose "Edit" on the Power Watchdog device from the ESPHome main UI screen.
10. I then copy/past the yaml file from the URL that is provided in the install guide, changing my MAC address to match my device.
11.  "Save" is clicked, then "Install" is clicked.
12.  When prompted about how to install Power Watchdog.yaml, I choose "Manual Download".
13.  On the next prompt, I chose "Modern Format".
14.  It begins to install and compile, but always fails at the firmware.elf linking line (see below).

Have I missed some steps here?  I also tried to do the install using the "Wireless" method, but it ends with the same results.  

Linking /data/hughes-power-watchdog/.pioenvs/hughes-power-watchdog/firmware.elf
collect2: fatal error: ld terminated with signal 9 [Killed]
compilation terminated.
*** [/data/hughes-power-watchdog/.pioenvs/hughes-power-watchdog/firmware.elf] Error 1
========================= [FAILED] Took 760.92 seconds =========================

tango2590, 2023-03-25:
Looks to be related to memory. The Hughes devices send a lot of data, and sometimes the ESP devices can't handle the incoming data plus the installation at the same time. My recommendation is to connect to your Hughes device using your phone before installation, and maintain the connection throughout the installation device. Since the Hughes only sends data to a single device, this will stop the incoming data to the ESP. Once the install is complete, you can disconnect your phone to allow the ESP to connect.

mllockwood, 2023-03-25:
Thanks, I’ll give that a try.  I assume the steps I’ve listed are correct
for the installation?--

Michael "Lee" Lockwood
***@***.***
281-713-0405

tango2590, 2023-03-25:
I'm not completely sure about 13 since that wasn't a thing when I added mine, but everything else seems correct. My rig is closed up for the winter still so I can't quite test it. Based on the output you posted, I'm fairly confident the ESP ran out of memory. It shouldn't take 760 seconds. In fact, it really shouldn't take more than a minute or so to install.

mllockwood, 2023-03-25:
I left my iPhone connected to the device, with the app open the entire time during the install process.  Unfortunately, the I have the same results:

Linking /data/hughes-power-watchdog/.pioenvs/hughes-power-watchdog/firmware.elf
collect2: fatal error: ld terminated with signal 9 [Killed]
compilation terminated.
*** [/data/hughes-power-watchdog/.pioenvs/hughes-power-watchdog/firmware.elf] Error 1
========================= [FAILED] Took 949.64 seconds =========================

spbrogan, 2023-03-25:
From your output it looks like your device is failing to compile.  I don't think this has anything to do with running out of memory on the esp.  When i follow your steps (which look good to me) I get the output below.  Linking is still part of the build process which is done on your HA server.  When you manually download you build the image on your HA server and then download the final esphome image.  If you are getting the file to download this is all still on the HA server and doesn't yet have anything to do with the esp.  

What type of system are you running for HA?  

```
Archiving /data/test_for_tango/.pioenvs/test_for_tango/libFrameworkArduino.a
Linking /data/test_for_tango/.pioenvs/test_for_tango/firmware.elf
RAM:   [==        ]  16.3% (used 53408 bytes from 327680 bytes)
Flash: [========= ]  90.6% (used 1661909 bytes from 1835008 bytes)
Building /data/test_for_tango/.pioenvs/test_for_tango/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
esp32_create_combined_bin(["/data/test_for_tango/.pioenvs/test_for_tango/firmware.bin"], ["/data/test_for_tango/.pioenvs/test_for_tango/firmware.elf"])
Wrote 0x1a7260 bytes to file /data/test_for_tango/.pioenvs/test_for_tango/firmware-factory.bin, ready to flash to offset 0x0
========================= [SUCCESS] Took 39.54 seconds =========================
INFO Successfully compiled program.
```

mllockwood, 2023-05-19:
I forgot to reply earlier.  I got this issue resolved.  I used a different instance of Home Assistant and was able to install via USB.

closed 2023-06-05 by tango2590