tango2590/Hughes-Power-Watchdog #9: Error when updating board after ESPHome 2023.8.1 update

mllockwood, 2023-08-18:
I am getting errors after updating to ESPHome 2023.8.1.  Primarily, this error repeatedly:

[esp32_ble:178]: (BLE) gap_event_handler - 3

Log files:

INFO ESPHome 2023.8.1
INFO Reading configuration /config/esphome/powerwatchdog.yaml...
INFO Updating https://github.com/spbrogan/esphome@PolledSensor
INFO Starting log output from hughes-power-watchdog.local using esphome API
INFO Successfully connected to hughes-power-watchdog.local
[14:19:15][I][app:102]: ESPHome version 2023.8.1 compiled on Aug 18 2023, 13:57:56
[14:19:15][C][wifi:543]: WiFi:
[14:19:15][C][wifi:379]:   Local MAC: C8:F0:9E:F3:D3:C0
[14:19:15][C][wifi:380]:   SSID: [redacted]
[14:19:15][C][wifi:381]:   IP Address: 10.10.0.42
[14:19:15][C][wifi:383]:   BSSID: [redacted]
[14:19:15][C][wifi:384]:   Hostname: 'hughes-power-watchdog'
[14:19:15][C][wifi:386]:   Signal strength: -39 dB ▂▄▆█
[14:19:15][V][wifi:388]:   Priority: -2.0
[14:19:15][C][wifi:390]:   Channel: 1
[14:19:15][C][wifi:391]:   Subnet: 255.255.255.0
[14:19:15][C][wifi:392]:   Gateway: 10.10.0.1
[14:19:15][C][wifi:393]:   DNS1: 10.10.0.1
[14:19:15][C][wifi:394]:   DNS2: 0.0.0.0
[14:19:15][C][logger:301]: Logger:
[14:19:15][C][logger:302]:   Level: VERBOSE
[14:19:15][C][logger:303]:   Log Baud Rate: 115200
[14:19:15][C][logger:305]:   Hardware UART: UART0
[14:19:15][C][logger:309]:   Level for 'esp32_ble_tracker': ERROR
[14:19:15][C][template.text_sensor:020]: Template Sensor 'Watchdog Monitor MAC Address'
[14:19:15][C][template.text_sensor:020]:   Icon: 'mdi:fingerprint'
[14:19:15][C][ble_switch:068]: BLE Client Switch 'Watchdog Monitoring'
[14:19:15][C][ble_switch:070]:   Icon: 'mdi:bluetooth'
[14:19:15][C][ble_switch:091]:   Restore Mode: always OFF
[14:19:15][C][hughes_power_watchdog:097]: Hughes Power Watchdog
[14:19:15][C][hughes_power_watchdog:098]:   Voltage Line 1 'Watchdog Voltage Line 1'
[14:19:16][C][hughes_power_watchdog:098]:     Device Class: 'voltage'
[14:19:16][C][hughes_power_watchdog:098]:     State Class: 'measurement'
[14:19:16][C][hughes_power_watchdog:098]:     Unit of Measurement: 'V'
[14:19:16][C][hughes_power_watchdog:098]:     Accuracy Decimals: 0
[14:19:16][C][hughes_power_watchdog:098]:     Icon: 'mdi:sine-wave'
[14:19:16][C][hughes_power_watchdog:099]:   Current Line 1 'Watchdog Current Line 1'
[14:19:16][C][hughes_power_watchdog:099]:     Device Class: 'current'
[14:19:16][C][hughes_power_watchdog:099]:     State Class: 'measurement'
[14:19:16][C][hughes_power_watchdog:099]:     Unit of Measurement: 'A'
[14:19:16][C][hughes_power_watchdog:099]:     Accuracy Decimals: 1
[14:19:16][C][hughes_power_watchdog:099]:     Icon: 'mdi:current-ac'
[14:19:16][C][hughes_power_watchdog:100]:   Power Line 1 'Watchdog Power Line 1'
[14:19:16][C][hughes_power_watchdog:100]:     Device Class: 'power'
[14:19:16][C][hughes_power_watchdog:100]:     State Class: 'measurement'
[14:19:16][C][hughes_power_watchdog:100]:     Unit of Measurement: 'W'
[14:19:16][C][hughes_power_watchdog:100]:     Accuracy Decimals: 0
[14:19:16][C][hughes_power_watchdog:100]:     Icon: 'mdi:lightning-bolt'
[14:19:16][C][hughes_power_watchdog:101]:   Voltage Line 2 'Watchdog Voltage Line 2'
[14:19:16][C][hughes_power_watchdog:101]:     Device Class: 'voltage'
[14:19:16][C][hughes_power_watchdog:101]:     State Class: 'measurement'
[14:19:16][C][hughes_power_watchdog:101]:     Unit of Measurement: 'V'
[14:19:16][C][hughes_power_watchdog:101]:     Accuracy Decimals: 0
[14:19:16][C][hughes_power_watchdog:101]:     Icon: 'mdi:sine-wave'
[14:19:16][C][hughes_power_watchdog:102]:   Current Line 2 'Watchdog Current Line 2'
[14:19:16][C][hughes_power_watchdog:102]:     Device Class: 'current'
[14:19:16][C][hughes_power_watchdog:102]:     State Class: 'measurement'
[14:19:16][C][hughes_power_watchdog:102]:     Unit of Measurement: 'A'
[14:19:16][C][hughes_power_watchdog:102]:     Accuracy Decimals: 1
[14:19:16][C][hughes_power_watchdog:102]:     Icon: 'mdi:current-ac'
[14:19:16][C][hughes_power_watchdog:103]:   Power Line 2 'Watchdog Power Line 2'
[14:19:16][C][hughes_power_watchdog:103]:     Device Class: 'power'
[14:19:16][C][hughes_power_watchdog:103]:     State Class: 'measurement'
[14:19:16][C][hughes_power_watchdog:103]:     Unit of Measurement: 'W'
[14:19:16][C][hughes_power_watchdog:103]:     Accuracy Decimals: 0
[14:19:16][C][hughes_power_watchdog:103]:     Icon: 'mdi:lightning-bolt'
[14:19:16][C][hughes_power_watchdog:104]:   Power Combined 'Watchdog Power'
[14:19:16][C][hughes_power_watchdog:104]:     Device Class: 'power'
[14:19:16][C][hughes_power_watchdog:104]:     State Class: 'measurement'
[14:19:16][C][hughes_power_watchdog:104]:     Unit of Measurement: 'W'
[14:19:16][C][hughes_power_watchdog:104]:     Accuracy Decimals: 0
[14:19:16][C][hughes_power_watchdog:104]:     Icon: 'mdi:lightning-bolt'
[14:19:16][C][hughes_power_watchdog:105]:   Cumulative Energy 'Watchdog Cumulative Power Usage'
[14:19:16][C][hughes_power_watchdog:105]:     Device Class: 'energy'
[14:19:16][C][hughes_power_watchdog:105]:     State Class: 'total_increasing'
[14:19:16][C][hughes_power_watchdog:105]:     Unit of Measurement: 'kWh'
[14:19:16][C][hughes_power_watchdog:105]:     Accuracy Decimals: 1
[14:19:16][C][hughes_power_watchdog:105]:     Icon: 'mdi:lightning-bolt-circle'
[14:19:16][C][hughes_power_watchdog:106]:   Error Code Value 'Watchdog Error Code'
[14:19:16][C][hughes_power_watchdog:106]:     State Class: ''
[14:19:16][C][hughes_power_watchdog:106]:     Unit of Measurement: ''
[14:19:16][C][hughes_power_watchdog:106]:     Accuracy Decimals: 0
[14:19:16][C][hughes_power_watchdog:106]:     Icon: 'mdi:alert-circle'
[14:19:16][C][hughes_power_watchdog:107]:   Error Text String 'Watchdog Error Text'
[14:19:16][C][hughes_power_watchdog:107]:     Icon: 'mdi:tooltip-text'
[14:19:16][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[14:19:16][C][version.text_sensor:021]: Version Text Sensor 'Watchdog ESPHome Version'
[14:19:16][C][version.text_sensor:021]:   Icon: 'mdi:new-box'
[14:19:16][V][version.text_sensor:021]:   Unique ID: 'c8f09ef3d3c0-version'
[14:19:16][C][esp32_ble:238]: ESP32 BLE:
[14:19:16][C][esp32_ble:240]:   MAC address: C8:F0:9E:F3:D3:C2
[14:19:16][C][esp32_ble:241]:   IO Capability: none
[14:19:16][C][ble_client:027]: BLE Client:
[14:19:16][C][ble_client:028]:   Address: 0C:61:CF:32:4B:81
[14:19:16][C][captive_portal:088]: Captive Portal:
[14:19:16][C][mdns:112]: mDNS:
[14:19:16][C][mdns:113]:   Hostname: hughes-power-watchdog
[14:19:16][V][mdns:114]:   Services:
[14:19:16][V][mdns:116]:   - _esphomelib, _tcp, 6053
[14:19:16][V][mdns:118]:     TXT: version = 2023.8.1
[14:19:16][V][mdns:118]:     TXT: mac = c8f09ef3d3c0
[14:19:16][V][mdns:118]:     TXT: platform = ESP32
[14:19:16][V][mdns:118]:     TXT: board = esp32dev
[14:19:16][V][mdns:118]:     TXT: network = wifi
[14:19:16][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[14:19:16][C][ota:093]: Over-The-Air Updates:
[14:19:16][C][ota:094]:   Address: hughes-power-watchdog.local:3232
[14:19:16][C][ota:097]:   Using Password.
[14:19:16][C][api:138]: API Server:
[14:19:16][C][api:139]:   Address: hughes-power-watchdog.local:6053
[14:19:16][C][api:143]:   Using noise encryption: NO
[14:19:16][C][wifi_info:009]: WifiInfo IPAddress 'Watchdog Host IP Address'
[14:19:16][C][wifi_info:009]:   Icon: 'mdi:fingerprint'
[14:19:16][V][wifi_info:009]:   Unique ID: 'c8f09ef3d3c0-wifiinfo-ip'
[14:19:16][C][wifi_info:011]: WifiInfo SSID 'Watchdog Host SSID'
[14:19:16][C][wifi_info:011]:   Icon: 'mdi:router-wireless'
[14:19:16][V][wifi_info:011]:   Unique ID: 'c8f09ef3d3c0-wifiinfo-ssid'
[14:19:17][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[14:19:17][V][hughes_power_watchdog:344]: Update Called
[14:19:17][V][hughes_power_watchdog:347]: Not Established

tango2590, 2023-10-07:
Are you still experiencing this issue?

mllockwood, 2023-10-08:
I reflashed the device and am not having issues anymore.

closed 2023-10-18 by tango2590