tango2590/Hughes-Power-Watchdog #12: Scanning for BLE I only see a device that is prefixed with PWS instead of PWD

philippelaurent, 2024-06-19:
I replaced the PWD30-EPO-H recently, and when scanning for BLE I only see a device that is prefixed with PWS instead of PWD. I know it's the right device because active connections interrupting the connection to another monitoring device can be performed easily. Unfortunately usable data cannot be retrieved from the PWD30-EPO-H. I can talk to it, but the only data retrieved is NaN. The previous PWD30-EPO-H returned usable information.

phurth, 2024-06-20:
You probably already know this, but if the EPO is still connected to the app on your phone, you'll keep getting NaN responses from the device. I knew this, but forgot and spent some time trying to figure it out before remembering I'd connected from my phone. Force-quitting the phone app fixed it.

aeronautjohn, 2024-07-01:
Has anyone successfully sniffed the MAC address using an iOS device? I'm reading conflicting reports about whether this is actually even possible with some suggesting iOS does some funny business to occlude or randomize the MAC address. I downloaded a few apps and none appeared to show a MAC address anywhere that I could find. But there's a higher than 0 chance I'm just an idiot. Anyone have some wisdom?

phurth, 2024-07-01:
I had the same problem and found the same answer - Apple does something on iOS (and Mac) devices to obfuscate bluetooth low energy device MAC addresses for security reasons. I had an Android device handy and was able to use that to get the address from my Watchdog.

aeronautjohn, 2024-07-03:
Yeah— I solved it by actually just using home assistant!

If you enable the Bluetooth LE Tracker integration, you just need to then open your known_devices.yaml file in your config tracker, and you’ll find the PWS/PWD device and MAC address. 

I actually did this entire integration / installation today in my camper, with the PWD connected to an outlet inside the camper running off of the inverter (with an adapter), all from an iPad. Even used the Raspberry Pi in my camper that runs home assistant as the device to program the ESP device. 

Still not entirely sure what that intermediary does or why it’s needed; instead of just using the HA servers onboard Bluetooth. But hey; it works!

Just finished setting up an automation that will change the metrics displayed on my main dashboard based on whether or not the PWD is present. Really fun stuff!

tango2590, 2024-07-04:
> If you enable the Bluetooth LE Tracker integration, you just need to then open your known_devices.yaml file in your config tracker, and you’ll find the PWS/PWD device and MAC address.

This is a great workaround! I'll add it to the readme to help others. I would've never known since I have an Android.

closed 2025-04-21 by philippelaurent