tango2590/Hughes-Power-Watchdog #13: Is it possible to do this with the onboard BLE of a Raspberry Pi 5?

aeronautjohn, 2024-07-03:
This is really more to broaden my understanding of these things than anything. 

What is the purpose of the ESP32 device acting in the middle? Is there a reason the Pi's onboard bluetooth (or whatever other machine is running Home Assistant) can't be used?

phurth, 2024-07-03:
I think the biggest reason is flexibility in locating the ESP32. Bluetooth has limited range, and if your HA server with Bluetooth is far from the Watchdog, the connection will fail or be unreliable.

aeronautjohn, 2024-07-03:
As it happens, the ESP32 is going to be installed right next to the
Raspberry Pi. The Pi itself is located up in a cabinet against the exterior
wall towards the back. That would be the ideal place for any device
attempting to connect to the Power Watchdog (just be sheer coincidence).
Though keeping it as high as possible is an intentional choice to improve
performance with other bluetooth sensors.

So just for the sake of simplicity; is it POSSIBLE to run this integration
without an ESP32? Since I won't be able to install one in any better
location than the Pi itself; the ESP32, I was planning to install directly
next to the Pi. I've got three of them not doing anything; but it would be
a nicer install if I could eliminate one more device that might not be
strictly necessary.

tango2590, 2024-07-04:
Initially, the ESP32 was used as the middleman because the BLE integration a few years ago was in its infancy. Plus, not everyone runs their HA on a Raspberry Pi. Now that the integration has been beefed up and is a valid alternative, to answer your question, I don't see why it couldn't work to eliminate the ESP device.

With that said, the Watchdog sends A LOT of data, to the point where the ESP32's memory is at like 80-90%. That's why it's recommended to not run anything else on it. I would wager a guess that using your Pi that's already running HA would run significantly slower if you were capturing BLE data from the Watchdog.

aeronautjohn, 2024-07-04:
Very cool!I’m running on an RPi 5 8GB; I would imagine I’ve got plenty of overhead even for that level of data. I’ve got it running right now using an ESP32; but what might I need to do to get it to run natively using the Pi’s integrated BLE controller? This is more an idle curiosity than anything at this point. I’m trying to broaden my knowledge of this stuff. Eventually, I’d like to tackle the project of sniffing out and controlling my LCI OneControl system via Bluetooth; if it’s even possible. Sent from my iPhone

tango2590, 2024-07-11:
Right now, the code is written for use in ESPHome to be deployed on devices such as an ESP32. I'm not 100% certain if it could be done using the integrated bluetooth controller on the PI, but I'm sure the probability is nonzero. I honestly don't know much about where the HA Bluetooth stuff is at right now. But even if you could use the onboard bluetooth sensor, a custom component/integration would need to be written to capture that data and turn it into individual sensors. And unfortunately, that's well outside of my wheelhouse.

aeronautjohn, 2024-07-11:
Gotcha!Yeah just a genuine question. Still learning about all of this and how it works. Of course it works just fine with an ESP32!Sent from my iPhone