tango2590/Hughes-Power-Watchdog #17: Error updating ESP32 with ESPHome 2025.7.1

twister36, 2025-07-17:
Error updating ESP32 with ESPHome 2025.7.1

Error Log:
```
> Error: The program size (1958668 bytes) is greater than maximum allowed (1835008 bytes)
> 
> 💡 TIP: Your ESP32 with Arduino framework has run out of flash space.
> 
> To fix this, switch to the ESP-IDF framework which is more memory efficient:
> 
> 1. In your YAML configuration, modify the framework section:
> 
>    esp32:
>      framework:
>        type: esp-idf
> 
> 2. Clean build files and compile again
> 
> Note: ESP-IDF uses less flash space and provides better performance.
> Some Arduino-specific libraries may need alternatives.
> 
> *** [checkprogsize] Explicit exit, status 1
> ========================= [FAILED] Took 11.20 seconds =========================
```

Following suggestion to change to ESP-IDF fails as well.

phurth, 2025-07-21:
Have you tried doing a clean build before installing?

[image]

phurth, 2025-07-21:
What board? On the latest ESPHome I'm running this along with a Victron shunt integration on the same nodemcu-32s board without issue.

bolducd, 2025-07-22:
With esphome 2025.7.1 the arduino library was updated to 3.0 and some components take more spaces on the device. Since its limited to 4mb its going over. You need to remove components you dont really need from the code. For me removing the logger and the wifi hotspot portal worked, i dont use them.

If you take this code dont forget to replace your bluetooth mac. My generator is not running so its not tested but it compiled and it's installed on my esp without error. 

esphome:
  name: hughes-power-watchdog

esp32:
  board: esp32dev
  framework:
    type: arduino

logger:
    
external_components:
  - source:
      type: git
      url: https://github.com/spbrogan/esphome
      ref: PolledSensor
    components: [ hughes_power_watchdog ]
    refresh: 1s

api:

ota:
  platform: esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

esp32_ble_tracker:

ble_client:
  - mac_address: REPLACE-ME-WITH-BLUETOOTH-MAC
    id: power_watchdog
    
switch:
  - platform: ble_client
    ble_client_id: power_watchdog
    name: "Watchdog Monitoring"

sensor:
  - platform: hughes_power_watchdog
    ble_client_id: power_watchdog
    voltage_line_1:
      name: "Watchdog Voltage Line 1"
      accuracy_decimals: 0
    current_line_1:
      name: "Watchdog Current Line 1"
      accuracy_decimals: 1
    power_line_1:
      name: "Watchdog Power Line 1"
      accuracy_decimals: 0
    voltage_line_2:
      name: "Watchdog Voltage Line 2"
      accuracy_decimals: 0
    current_line_2:
      name: "Watchdog Current Line 2"
      accuracy_decimals: 1
    power_line_2:
      name: "Watchdog Power Line 2"
      accuracy_decimals: 0
    combined_power:
      name: "Watchdog Power"
      accuracy_decimals: 0
    total_power:
      name: "Watchdog Cumulative Power Usage"
      accuracy_decimals: 1
    error_code_value:
      name: "Watchdog Error Code"
    error_code_text:
      name: "Watchdog Error Text"
   
text_sensor:
  - platform: version
    name: Watchdog ESPHome Version
    id: watchdog_esphome_version
  - platform: wifi_info
    ip_address:
      name: Watchdog Host IP Address
      id: watchdog_host_ip
      icon: mdi:fingerprint
    ssid:
      name: Watchdog Host SSID
      id: watchdog_host_ssid
      icon: "mdi:router-wireless"
  - platform: template
    id: watchdog_host_mac_address
    name: Watchdog Monitor MAC Address
    lambda: 'return {WiFi.macAddress().c_str()};'
    icon: mdi:fingerprint
    update_interval: 1d

twister36, 2026-02-12:
> FYI, I recently used the reverse engineering that was done for this project to create a HomeAssistant integration that uses bluetooth proxy rather than directly putting it on the ESP. It works for my PWD50-EPD (GEN I). Try it out and see if it works for you and provide feedback. https://github.com/john-k-mcdowell/My-Hughes-Power-Watchdog

Thank you... I just came back here to look for updates, as I need to change my wifi network and can't do that either through this ESPHome due the issue I reported above. 

I will give your HACS version a try!